Team-Neptune/DeepSea #9: deepsea1.3.0-patch package still has issue in sysnand cfw

paulzheng136, 2020-06-02:
deepsea1.3.0-patch package still has issue in sysnand cfw
thanks TN release aio deepsea  package.  i found a little issue when i use it. 
if you want to hack sysnand,  some of nsp can't run in cfw-sysnand,   i assume maybe the problem is patch.ini,  it's so weired that nothing happened in cfw-emummc .   so i add a line in hekate_ipl.ini to fix it as follows :
1, i DL loader.kip from haruko's github page and  copy to sd:atmosphere/kips/ 's folder.
2,   i add a line as "kip1=atmosphere/kips/loader.kip" below "kip1=nosigchk" in [cfw(sysnand)].

the same issue also with kosmos 16.1 package after i tested it.

paulzheng136, 2020-06-02:
[config]
autoboot=0
autoboot_list=0
bootwait=1
backlight=100
autohosoff=0
autonogc=1

{}
{www.91wii.com}
{}

{--- Custom Firmware ---}
[DS-CFW (SYSNAND)]
fss0=atmosphere/fusee-secondary.bin
kip1patch=nosigchk
kip1=atmosphere/kips/loader.kip
atmosphere=1
emummc_force_disable=1
icon=bootloader/res/icon_payload.bmp
{}

[DS-CFW (EMUMMC)]
fss0=atmosphere/fusee-secondary.bin
kip1patch=nosigchk
emummcforce=1
atmosphere=1
cal0blank
icon=bootloader/res/icon_payload.bmp
{}

{--- Stock ---}
[NO-CFW(SYSNAND)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
stock=1
icon=bootloader/res/icon_switch.bmp
{}

hax4dazy, 2020-06-02:
I haven't had problems running NSP's in sysnand. What issues did you see?

paulzheng136, 2020-06-02:
not all nsps.  some of nsps. for example  tinfoil 8.1's old version which support running in kosmos.  
it shows  unable to start software.  return to the home menu and try again.

paulzheng136, 2020-06-02:
Hos 10.0.3 both  sysnand and emummc,   i change the theme color in emummc.  so emunand is correct.   i copied deepsea 1.3.0 patched package in SD card.
so could  you try to install different nsps, especially choose some old games.

hax4dazy, 2020-06-02:
We do not support tinfoil

marco-calautti, 2020-06-02:
This has nothing to do with how deepsea packages patches, I believe. If an issues exists, it is either because of HarukoNX' patches, or (most probably) Hekate patching system. Again, the only safe way to apply patches to your CFW is by booting via fusee-primary.bin (fusee patches are also included in deepsea).

paulzheng136, 2020-06-02:
> We do not support tinfoil
i ran tinfoil in deepsea well after add loader.kip in sysnand.  but in emummc works good without changing.

marco-calautti, 2020-06-02:
Loader.kip is nothing else that loader patches in patches.ini packaged in a kip for atmosphere. Everything should work fine with it in both sys cfw and emu cfw. If this is not the case, then Hekate is not applying patches properly. As I said, try booting with fusee-primary.bin to sysnand cfw and see if your nsps work fine. If this is the case, the issue is with hekate.

paulzheng136, 2020-06-02:
> This has nothing to do with how deepsea packages patches, I believe. If an issues exists, it is either because of HarukoNX' patches, or (most probably) Hekate patching system. Again, the only safe way to apply patches to your CFW is by booting via fusee-primary.bin (fusee patches are also included in deepsea). Did you try booting syscfw with fusee-primary.bin and check that your nsps work?
yes, you are right.  deepsea1.3.0-patched contains fs patches and loader patches in atmosphere/kip_patches/ even it's useless  if you choose fss0 to boot fusee-seconday.bin
if i want to payload fusee-primary.bin.  i have to add a extra line in hekate_ipl.ini to run.

marco-calautti, 2020-06-02:
> yes, you are right. deepsea1.3.0-patched contains fs patches and loader patches in atmosphere/kip_patches/ even it's useless if you choose fss0 to boot fusee-seconday.bin
> if i want to payload fusee-primary.bin. i have to add a extra line in hekate_ipl.ini to run.

No, you don't need anything like that. You just need to push fusee-primary.bin directly with your RCM payload injector, and forget about Hekate for a moment.

paulzheng136, 2020-06-02:
> Loader.kip is nothing else that loader patches in patches.ini packaged in a kip for atmosphere. Everything should work fine without loader.kip in both sys cfw and emu cfw. If this is not the case, then Hekate is not applying patches properly. As I said, try booting with fusee-primary.bin to sysnand cfw and see if your nsps work fine. If this is the case, the issue is with hekate.

> No, you don't need anything like that. You just need to push fusee-primary.bin directly with your RCM payload injector, and forget about Hekate for a moment.

well.  i think if i add a extra line like  
[atmosphere]
payload=bootloader/payloads/fusee-primary.bin
thant's more convienience to launch. lol.

marco-calautti, 2020-06-02:
> well. i think if i add a extra line like
> [atmosphere]
> payload=bootloader/payloads/fusee-primary.bin
> thant's more convienience to launch. lol.

There have been reports stating that even in that way, sometimes patches might not work. pushing fusee-primary,bin directly is the only safe way to be sure that patches are working on sysnand.

hax4dazy, 2020-06-02:
This isn't related to DeepSea any more. Please move this conversation somewhere else

paulzheng136, 2020-06-02:
> This isn't related to DeepSea any more. Please move this conversation somewhere else
i just suggest that deepsea can add loader.kip in patched pageage.

paulzheng136, 2020-06-02:
> > yes, you are right. deepsea1.3.0-patched contains fs patches and loader patches in atmosphere/kip_patches/ even it's useless if you choose fss0 to boot fusee-seconday.bin
> > if i want to payload fusee-primary.bin. i have to add a extra line in hekate_ipl.ini to run.
> 
> No, you don't need anything like that. You just need to push fusee-primary.bin directly with your RCM payload injector, and forget about Hekate for a moment. If it boots to emummc, open the emummc.ii file and change "enabled=1" to "enabled=0", shutdown and push fusee-primary.bin again.

> Loader.kip is nothing else that loader patches in patches.ini packaged in a kip for atmosphere. Everything should work fine without loader.kip in both sys cfw and emu cfw. If this is not the case, [...]

i tested it before,  it works well if i payload fusee-primary.bin directly.  so that's why the issue is in patch.ini  not deepsea.  but i just suggest to add one more kip loader in cfw-sysnand

Slluxx, 2020-06-02:
Tinfoil does not support fss0=fusee-secondary booting. thats a personal desicion made by blawar and nothing we can or will fix.

also, we do not support payload=fusee-primary booting and thus this issue gets closed now.
If you want some kind of discussion about that, move to discord please.

closed 2020-06-02 by Slluxx