Team-Neptune/DeepSea #14: SSL connect error

nadrino, 2020-06-27:
Hello there :)

Each time I launch the updater I get this error:

[image]

Any idea from where that could come from ?

hax4dazy, 2020-06-27:
Did you use Incognito?

nadrino, 2020-06-27:
Yes I did use IncognitoRCM to remove my prod info, shouldn't I ?

hax4dazy, 2020-06-27:
Yes you should not use it. It fucked over your SSL sert. Do you have a backup that you can restore to get your prodinfo back?

nadrino, 2020-06-27:
Yes I have a backup of my prodinfo, but I used Incognito to avoid getting banned on OFW and still get the updates of my games.

Is there a way to run the updater without SSL ?

hax4dazy, 2020-06-27:
No. You can prevent getting banned by using 90's dns or atmosphere's own "incognito"

nadrino, 2020-06-27:
Alright, I'll check that out. Is there a reason why you have to use SSL to connect whereas other homebrew don't need it ? If that's not absolutely necessary, you could let us dl update and put a warning message about this issue

hax4dazy, 2020-06-27:
Github requires SSL.

closed 2020-06-27 by Slluxx

Ninjistix, 2020-08-17:
Was also having this issue, other updaters via github work, but this is the only app that gives the SSL error connection.

hax4dazy, 2020-08-17:
> Was also having this issue, other updaters via github work, but this is the only app that gives the SSL error connection.

You have then used incognito to ""wipe"" your prodinfo. This also wiped your ssl cert. Don't use incognito. If you have a backup restore it and only use 90s dns

TechGeekGamer, 2020-08-18:
* 90 DNS