Team-Neptune/DeepSea #22: DeepSea Updater Crashes Switch

BOsborneOfficial, 2020-10-27:
When I update DeepSea from the updater homebrew app atmosphere crashes and I have to restart. I will put my FW version, DeepSea Version, Atmosphere Version, and DeepSea Updater Version in the morning when I have access to my pc. I am willing to help fix the issue and if I find a fix I will send a pull request. (This has probably been fixed but I never saw an issue created so I decide to create one myself)

Slluxx, 2020-10-27:
weird. it never crashed when we tried it. if your switch restarts, is deepsea updated then? maybe you are updating form a very old version?

can you try installing the second newest deepsea version and then updating via the updater?

hax4dazy, 2020-10-27:
Are you running it in applet mode? If yes then this is a bug that's already known. You'd have to run the app with full ram access to have it not crash.

BOsborneOfficial, 2020-10-27:
> weird. it never crashed when we tried it. if your switch restarts, is deepsea updated then? maybe you are updating form a very old version?
> 
> 
> 
> can you try installing the second newest deepsea version and then updating via the updater?

When the switch crashes the download bar is at about 75 - 90%. When the switch turns back on DeepSea is not updated.

> Are you running it in applet mode? If yes then this is a bug that's already known. You'd have to run the app with full ram access to have it not crash.

Yes and no, but sometimes when I use anything on full ram access the switch crashes so I have not tried the Updater app. I will try that right now, thanks!

BOsborneOfficial, 2020-10-27:
Ok found out I had a bad theme installed. So once I uninstalled my theme the hb-menu would work with full ram access and I was able to update to the newest version. P.s. I like the Nintendo/DeepSea boot logo! 👍 if I can contribute in any way I will be more than happy to!

closed 2020-10-27 by BOsborneOfficial