Team-Neptune/DeepSea #25: Tesla Overlay doesn't work

airchase, 2020-10-30:
Tesla is enabled in the DeepSea toolbox, however when holding **Down L** + **Down D Pad** + **Down Joystick** nothing happens at all.

TechGeekGamer, 2020-10-30:
By "Down Joystick", I'm assuming you mean clicking it, right? Correct me if I'm wrong please.

airchase, 2020-11-02:
Oops, I'm stupid sorry. My only really indicator of how Tesla Overlay was actually launched was in a YouTube video, and he said "down joystick", so my bad.

Not really necessary, but if there a was a quick FAQ / guide quickly highlighting how certain features work, and what apps do what, that might be helpful for people who aren't as involved in the Switch homebrew scene. Case in point, I just found out there's a way to transfer files to the SD card over USB without taking the card out with nxmtp, which is awesome! It wasn't until I Google'd if there was a way to do it and tried to install an app to accomplish it however did I find out I already had one :)

closed 2021-01-31 by Slluxx