Team-Neptune/DeepSea #39: Error : "failed to run sept, Main BCT is improper" after updating DeepSea Package

kaydayyy, 2021-02-10:
Hello, yesterday I updated the deepsea package and aftee booting up the sysnand or emunand I get that error, and I cant find any fix, what should I do? please help :(

hax4dazy, 2021-02-10:
Did you before updating DeepSea update your firmware? And please if you can join our [discord](https://discord.gg/VkaRjYN) server for better support

kaydayyy, 2021-02-10:
yes, before updating deepsea I did update my console firmware :(

hax4dazy, 2021-02-10:
How did you update the console firmware?

kaydayyy, 2021-02-10:
I updated via console settings, using internet..., and then I created my emunand

Slluxx, 2021-02-10:
This is from Hekate 5.5.4's changelogs:

```
Main BCT in BOOT0 is now checked for improper values before running Sept in Erista
Before, launching Sept in such cases (improper BCT) it would make it black screen.
Now an error will show up that instructs user to run sept with proper BCT at least once, in order to cache keys in EKS.
```

looks like something is wrong with your boot0 partition

kaydayyy, 2021-02-10:
how can I fix it?

hax4dazy, 2021-02-10:
We recommend you join this [discord](https://discord.gg/ZdqEhed) server and ask for further info. Once we know how to fix it we will pin a solution in the Support channel in [our discord](https://discord.gg/VkaRjYN)

closed 2021-02-10 by hax4dazy