Team-Neptune/DeepSea #40: Can I write via ftp?

lesca, 2021-02-15:
I am using mac Finder to connect my switch via ftp://ip:5000
Listing and opening files are fine, but when I edit and save a fie (config.ini file), I got error message say no write access. 

Is it possible to enable the write permission for ftpd? I've went through the sys-ftpd/config.ini file, but find no option for this. If it's not supported by now, will it be supported in coming versions? Thanks !

Slluxx, 2021-02-15:
try to use filezilla. thats a pretty good and afaik the standart ftp tool. If it works with that tool, the issue is mac finder. if it doesnt work, we should take a closer look at ftpd

lesca, 2021-02-15:
@Slluxx you're right. I need a ftp client to transfer files to my switch! Now I made it work. Thanks.

closed 2021-02-15 by lesca