Team-Neptune/DeepSea #42: About /bootloader/payloads/ folder

chikenstew, 2021-02-25:
I noticed that since the 2.0.0 release, you no longer include the fusee-primary and Lockpick_RCM payloads in the /bootloader/payloads/ folder...Would you reconsider including those in future releases, since I find them to be rather useful and I am pretty sure there are many others who would as well? Thanks!

Slluxx, 2021-02-25:
We decided to remove fuse-primary from the payloads folder because we don't even support booting it. It was included because that was just how the old build script worked. Lockpick isn't needed for regular use either (which doesn't mean its not handy to have) thus we removed it just to shave off some bytes from the release. If you really need it, you can always get it from the "all-in-one switch updater" homebrew which we include in our releases.

Is there any particular reason why we should include it directly instead of the user downloading it separately from the included homebrew?

chikenstew, 2021-02-26:
I can understand how Lockpick is only used once after every time you upgrade to a major new system firmware (and therefore not considered as a "regular-usage" homebrew like the others), but I think most of us when we do upgrade to a new version of deepsea it's because we're also upgrading to a new firmware, and by not including it, I feel it makes deepsea less of an "all-in-one" package and less convenient.
BTW I have wifi turned off on my switch permanently (for fear of console banning), so the aio updater (which requires internet connection I believe) is not an option for me.

Slluxx, 2021-02-26:
I think ive used lockpick only once or twice in my life. It was to extract my console keys so i could use them in yuzu (nintendo switch emulator). Whats the reason for you to use lockpick after each major fw update?

We support 90DNS and even have our own Aquatic DNS service, that lets people play online whilst blocking some (but not all) connections to Nintendo. Our package is basically "designed" to use those services.

Yeah i get your point, especially if you have wifi turned off.
But whats the reason behind that? Just using a CFW will not ban you, even if you play online. Whats really bothering nintendo is piracy and modding online games. If you are afraid of a ban and don't go online for that reason, why not use 90DNS? Its purpose is to block all connections to nintendo. You could even do shady things and not get banned.

Since we don't support piracy but do support being online, your case is pretty much unique.

Slluxx, 2021-02-26:
@hax4dazy whats your opinion on this?

chikenstew, 2021-02-26:
> I think ive used lockpick only once or twice in my life. It was to extract my console keys so i could use them in yuzu (nintendo switch emulator). Whats the reason for you to use lockpick after each major fw update?
> 
I thought console keys get updated with new keys with each major new firmware?

> We support 90DNS and even have our own Aquatic DNS service, that lets people play online whilst blocking some (but not all) connections to Nintendo. Our package is basically "designed" to use those services.
> 

As you said, 90DNS doesn't block all nintendo connections...I don't like it when nintendo arbitrarily pushes system firmware/game updates onto my switch...I prefer having more control myself.

Anyways, if it's too much hassle for guys to reinstate those, then I guess it's OK...I can do it myself, but it just means more work on my part as a user.

hax4dazy, 2021-02-26:
> Another reason I prefer to keep my switch offline is I don't like it when nintendo arbitrarily pushes system firmware/game updates onto my switch...I prefer having more control myself.
> 
> I used to use 90DNS, but after reading stories on gbatemp of how some people still got pushed with nintendo updates even with 90DNS set up, I decided not to take the risk.

There is no way that Nintendo is able to reach you while you're using 90DNS but if you're still scared of having updates pushed to you you can always permanently enable 90DNS using the new atmosphere sysmodule.

I also see no reason to include Lockpic in our AiO pack. And as for including fusee-primary I don't see a reason why we would do that. Hekate is better as a payload and the only "real" reason to include fusee-primary is for a piracy tool that we don't support.

Slluxx, 2021-02-26:
It doesn't look like we will include those payloads again in the future, since 90DNS is a perfectly fine solution to go online and block all connections to Nintendo (even updates). 

I will leave this ticket open for some time and if more people respond, we will think about it again.

Slluxx, 2021-05-02:
Since there is no interest by other people, i'll close this now.

closed 2021-05-02 by Slluxx