Team-Neptune/DeepSea #43: The contents of "hekate_ipl.ini" disappear

ma2kaz, 2021-03-21:
I'm trying to put "Tegra Rcm GUI 2.6" and "deepsea-normal_v2.1.0.zip" into "switch 11.0.1".
However, even if I write the settings in "bootloader / hekate_ipl.ini", the contents disappear.
I don't know the cause. Please let me know.

--- The contents are as follows
[config]
autoboot=0
autoboot_list=0
bootwait=3
verification=1
backlight=100
autohosoff=0
autonogc=1
updater2p=1

{--- DeepSea ---}
{DeepSea}
{Github: https://github.com/orgs/Team-Neptune/}

{--- Custom Firmware ---}
[CFW (SYSNAND)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
atmosphere=1
logopath=bootloader/bootlogo.bmp
icon=bootloader/res/icon_payload.bmp
{}

[CFW (EMUMMC)]
emummcforce=1
fss0=atmosphere/fusee-secondary.bin
atmosphere=1
logopath=bootloader/bootlogo.bmp
icon=bootloader/res/icon_payload.bmp
{}

{--- Stock ---}
[Stock (SYSNAND)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
stock=1
icon=bootloader/res/icon_switch.bmp
{}

Slluxx, 2021-03-21:
I am not sure what you mean. How and why do you want to put TegraRCM GUI on your switch? Thats a program for a computer, not your switch. How are you editing the files contents and what do you do afterwards? How do you know those settings are not saved?

What exactly is your goal?

ma2kaz, 2021-03-21:
Excuse me.
"Tegra Rcm GUI 2.6" is used to "send payload" from Win10 to "switch".
Put "deepsea-normal_v2.1.0.zip" in the microSD (zip format is decompressed).
And even if I write the setting to "bootloader / hekate_ipl.ini", the contents disappear.

Sorry for the incomprehensible writing.
I hope you can understand it with the above explanation. Thank you.

Slluxx, 2021-03-21:
Can you tell me what kind of sdcard you have maker and model? And how do you put the contents on it?
Is the content there but disappears after you have edited it? What if you do not edit it and just try to boot it?

ma2kaz, 2021-03-22:
This is the SD card you are using.
Manufacturer: Vosiy
Memory type: Micro SDXC
Speed: UHS-I U3 Class 10
Capacity: 256GB

What does content mean?
If you tell me, I'll edit it. Thank you.

Slluxx, 2021-03-22:
Please take a look at this: https://flashfakecentral.wordpress.com/testing-flash-memory-chips/h2testw-english/
Download H2test2w to check if your sdcard is a fake card or a legit one. Ive never heard of "Voisy" so i just want to make sure its all legit on that part.

> What does content mean?

"[...] However, even if I write the settings in "bootloader / hekate_ipl.ini", the contents disappear."

Why do you actually want to write settings manually? The only thing you need to do is extract the contents of the zip file on your sd and thats it. you dont need to edit anything. Please just leave it as is and test it in your switch.

ma2kaz, 2021-03-24:
I'm sorry that the day is free.
I tried using "h2testw 1.4". The diagnosis stopped halfway.
When I used another SD card, the settings never disappeared.
Thank you, I was able to solve it.

Slluxx, 2021-03-24:
Just in case anyone has the same problem, the sdcard is probably fake. It works by modifying the sdcard in such a way that it displays any amount of gigabyte but its actual capacity is way less. When trying to copy something to the card, the whole copying process will look like normal but when the cards true capacity is used, the copying process will start at the beginning an overwrite the files that are already on the sd. For the consumer it will look like everything is working till they try to open/edit/save a file.

closed 2021-03-24 by Slluxx