Team-Neptune/DeepSea #44: After upgrade not able to enter into hbl menu

rkmax, 2021-04-13:
Hi, a couple of days ago I was running previous release of deepsea with firmware 11.0.1 everything was fine but then I left the wireless enabled then the console was updated to 12.0.0 I waited until the firmware 12.0.0 was supported now I upgrade as soon start the console, run hetake go to atmosphere I can see the boot logo of deepsea then the purple logo of atmosphere then a black/white and finally I see the original nintendo logo (not the modified logo from deepsea) any help you cloud provide to be able to run again the homebrew version?

hax4dazy, 2021-04-13:
Can you please join our [Discord server ](discord.gg/VkaRjYN) so we can better help you with this issue?

rkmax, 2021-04-13:
Surem thx!

closed 2021-04-13 by hax4dazy