Team-Neptune/DeepSea #50: panic occurred

miguel197843, 2021-04-22:
I couldn't load hekate. it sends a error message. Panic Occurred.... What's the problem?

Slluxx, 2021-04-22:
Are you using the included payload?

miguel197843, 2021-04-22:
Yes. hekate_ctcaer_5.5.5.bin

Slluxx, 2021-04-22:
did you update manually or through aio updater?

miguel197843, 2021-04-22:
Sorry. Update what? atmosphere?

miguel197843, 2021-04-22:
I put the files in the sd card, injected hekate payload and then I get the message panic occurred...

hax4dazy, 2021-04-22:
Can you give use more information then just "panic occurred"? Maybe join our [Discord Server](discord.gg/VkaRjYN) so we can help you better

miguel197843, 2021-04-22:
[image]

miguel197843, 2021-04-22:
Panic occurred while running atmosphere.
Id title: 010041544D530000
Error: std: : abort (0xffe)

hax4dazy, 2021-04-22:
Does it say that it saved a report?

miguel197843, 2021-04-22:
> Does it say that it saved a report?

Yes

hax4dazy, 2021-04-22:
Can you send it to me so I can take a look at it?

miguel197843, 2021-04-22:
> Can you send it to me so I can take a look at it?

Sure. How can I do that?

miguel197843, 2021-04-22:
[report_0645d289.bin.zip](https://github.com/Team-Neptune/DeepSea/files/6359493/report_0645d289.bin.zip)

miguel197843, 2021-04-22:
> Can you set the archive bit in hekate?

Sorry, I didn't understand. I'm not an expert. Did you figure out the problem?

hax4dazy, 2021-04-22:
Ok so do these steps:

1. Boot into hekate
2. Select tools (at the top)
3. Click on "arch bit * RCM * Touch * Pkg1/2 (at the bottom)
4. Click on fix archive bit
5. wait and see if it fixed anything and then boot your system

miguel197843, 2021-04-22:
> Ok so do these steps:
> 
> 1. Boot into hekate
> 2. Select tools (at the top)
> 3. Click on "arch bit * RCM * Touch * Pkg1/2 (at the bottom)
> 4. Click on fix archive bit
> 5. wait and see if it fixed anything and then boot your system

It worked! You're genius. Now I have another problem. The games don't work.

hax4dazy, 2021-04-22:
Did you pirate them?

miguel197843, 2021-04-22:
> Did you pirate them?

Someone did, not me. Just copied them to the sd card.

miguel197843, 2021-04-22:
Should I update sigpatches through Aio?

hax4dazy, 2021-04-22:
We will not provide support for pirated games. You will have to find out yourself how to make them work again.

hax4dazy, 2021-04-22:
Anyways seeing as this solved your issue I will close this Issue now. Feel free to join our [Discord Server](discord.gg/VkaRjYN) if you have any other questions

closed 2021-04-22 by hax4dazy