Team-Neptune/DeepSea #59: Issue: Deep Sea Will not boot CFW

quietmansoath, 2021-06-02:


TransKat, 2021-06-02:
You'll need to provide more information.

hax4dazy, 2021-06-02:
Delete your atmosphere, sept and bootloader folders, grab them from our latest release and put them onto your sd card

quietmansoath, 2021-06-02:
Okay. I'll try that again and report back. For the record, I'm coming from Kosmos so I had done all of that, but I'll do it once more just to make sure. Thank you.

hax4dazy, 2021-06-02:
Are you sending the latest hekate payload?

quietmansoath, 2021-06-02:
Yes; it's the one bundled with your latest release.

hax4dazy, 2021-06-02:
Can you try sending the payload thru a computer or phone instead of using the dongle?

quietmansoath, 2021-06-02:
Okay. Will take me longer since I don't have any of that configured and need to find apps etc to do that. I'll follow up later. Thank you.

quietmansoath, 2021-06-02:
So I ended up trying it with my phone and it still didn't work. I went back to Kosmos and things are working fine, but it did allow me to realize I should be more clear. It's not that the screen is going black after the Deep Sea splash screen; it's that it was turning off. So Hekate > emunand or sysnand boot > deep sea splash screen > switch turns off.

Slluxx, 2021-06-02:
1) are you trying to boot emunand or sysnand?
2) do you have any logfiles/reports of hekate or atmosphere after a clean installation and one run?
3) what is your sdcard formated in?
4) what is the folderstructure of your sdcard after manually installing deepsea? (provide image please)
5) did you delete `atmosphere`, `bootloader` & `sept` before installing deepsea?

Ammako, 2021-06-03:
Check first with clean atmosphere 0.19.3

carcaschoi, 2021-06-04:
Delete Deepsea bootlogo

Ammako, 2021-06-30:
@quietmansoath Any updates? Did you get anything working since?

There is always a possibility that newer versions of Atmosphere and/or Hekate could have incompatibilities with older firmware versions (has happened very recently in fact.) I think it's important to try with regular atmosphere and fusee-primary first just to rule out deepsea or hekate as the culprit.

If it *does* boot fine with plain atmosphere, *then* further troubleshooting can start to determine whether the issue is with hekate, or with something else that deepsea package does. It's possible you might simply need to update to a newer firmware version, but if you do, make a backup first, so you can restore back to a working state if anything goes wrong.

Slluxx, 2021-07-02:
closed due to inactivity. please join our discord if further help is needed

closed 2021-07-02 by Slluxx