Team-Neptune/DeepSea #62: [Feature request] Add 90dns host file options on the builder net

carcaschoi, 2021-07-08:


Slluxx, 2021-09-16:
will get added soon

ShadowOne333, 2021-11-06:
Here are mine in case it helps.
These go inside /atmosphere/hosts/

[default.txt](https://github.com/Team-Neptune/DeepSea/files/7491277/default.txt)
[sysmmc.txt](https://github.com/Team-Neptune/DeepSea/files/7491278/sysmmc.txt)

Slluxx, 2021-11-07:
We had a lenghty discussion about including them in our packages or not. 

The pro arguments are:
- it results in more security

The con arguments are:
- it will annoy people that go online with CFW
- it results in a false sense of security and a "apply it once, never care about it again" mentality

In the case of Nintendo adding new telemetry domains, people that think they are secure because they use a hostfile are actually not. Obviously the same applies to 90DNS but they are fast in figuring out what domains got added and their changes instantly apply to everyone using the DNS, whilst people that use hostfiles need to either manually add the new domains or update their package to block telemetry again. Since people like "apply and never care again", most of the users don't know when this happens or how to get/enable protection again which in the end will most likely cause more banned switches.

My personal opinion/solution would be to inform users more about the risks, make them understand the consequences and to not use hostfiles. It will raise awareness and DeepSea/Team Neptune is not and can not be held responsible for any banned switches this way (even though we couldnt be held responsible whatsoever, we dont like people blaiming us for misstakes - which happens way too much).

I will leave this ticket open for discussion. Maybe someone has a good idea about how to deal with this situation.

woolrage, 2021-11-13:
I think the only solution can be a hosts file that is auto-updated with all domains 90dns blocks... and to use both for added security. There can not be a "never care again" solution. Perhaps there could be an easy opt-out in DeepSea Toolbox for online users?

Slluxx, 2021-11-14:
Updating the hosts file will require a console reboot. All this can be done from a sysmodule but we can't just restart a console whenever we want to. A user may do critical tasks, like updating the firmware or installing a game. There is no way to inform a user without their action (eg we can't push notifications, the user has to manually click to view dynamic content somewhere - like in a homebrew or Tesla overlay). Thus an automatic update solution - unless a user can directly be notified somehow - is out of question. 

A different point would also be that we would have to maintain a repository with the hostfiles. That's not something we want to have the responsibility for.

lhvy, 2021-12-05:
> A different point would also be that we would have to maintain a repository with the hostfiles. That's not something we want to have the responsibility for.

90DNS has a repo on GitLab that may have all the hosts in a format we could use. I know many other AIO switch CFW packs include the hosts files to block Nintendo, maybe these files could just be made easily available somewhere on this repo rather than including them in DeepSea directly.

Regarding auto-updates, I assume that changes to Nintendo telemetry links would happen between firmware updates. As such, auto-updates aren't really required provided that the user updates their hosts file before a system update (which would also lead to a reboot, fixing the reboot problem) maybe a hosts updater could be added to DeapSea and users advised to update hosts before updating their CFW firmware? Then again, maybe something like this is out of scope for DeepSea is more suited for aio-switch-updater.

hax4dazy, 2023-10-29:
Closing this as done and completed

closed 2023-10-29 by hax4dazy