Team-Neptune/DeepSea #63: Missioncontrol

VDDZ, 2021-07-12:
Cannot use missioncontrol. If I delete the contents title id folder, it boots fine, but if I reinstall, it crashes on the splash screen, no atmosphere crash screen either. Hope this is where to post this issue...

hax4dazy, 2021-07-12:
Hey! This repository is only for issues with the DeepSea build script, please join our [Discord Server](discord.gg/VkaRjYN) for getting support for your switch.

Thanks!
- Team Neptune

Slluxx, 2021-07-12:
AiO-updater and our builder website do not offer MissionControl. How exactly did you install it?

MissionControl had some issues in the past with the newer atmosphere and switch firmwares. It got updated 12 hours ago but requires a minimum atmosphere version. Make sure you meet the requirements and do not have too many sysmodules running.

VDDZ, 2021-07-12:
HamletDuFromage's aio app, then did a new custom package build, picked missioncontrol. Reboot took me to hekate, tried to boot primary, then fail. Think that's the simplest way to describe the process.

Slluxx, 2021-07-12:
@hax4dazy Can you recheck that please?

VDDZ, 2021-07-12:
Yeah, gonna do a fresh fresh install tomorrow, will report back. Thx for the fast reply

hax4dazy, 2021-07-12:
I wasn't able to find mission control in AiO updater

VDDZ, 2021-07-12:
It's not there... My mistake, swore it was. New release of missioncontrol 5.0.0 runs, thanks all

closed 2021-07-12 by VDDZ