Team-Neptune/DeepSea #70: Support For HOS 13 & Atmosphere 1.1.0

idarkside, 2021-09-22:
I know it has only just been released but do you plan to add support for HOS 13 and atmosphere 1.1.0?

Thanks for the brilliant project

Slluxx, 2021-09-22:
Atmospheres 13.0.0 support is in pre-release state. We will not update DeepSea as long as its not "officially released".

takoyaki99, 2021-09-25:
https://github.com/Atmosphere-NX/Atmosphere/releases/tag/1.1.1
Official release out, I think,

jonahnm, 2021-09-25:
> Atmospheres 13.0.0 support is in pre-release state. We will not update DeepSea as long as its not "officially released".

it's officially released now.

Slluxx, 2021-09-25:
we still have a waiting period on any release. apart from that, we have a reallife too.
be patient.

idarkside, 2021-09-25:
Sorry guys didn't want to open a can of worms, thanks for the great work, i will close this issue 👍

closed 2021-09-25 by idarkside