Team-Neptune/DeepSea #72: Downloading new deep sea update

Jfowler30, 2021-09-27:
Hello I am trying to download the new deep sea update but it keeps coming up with an error code of 2349-008 picture attached as well thanks
[image]

Jfowler30, 2021-09-27:
Update: managed to get passed error but now I have a black screen and nothing happens, obviously I did something wrong 😅

Slluxx, 2021-09-27:
well, the title id is for ARMS, which doesnt help a lot.
the error code stands for "LibnxBinderError_NoInit". I assume the homebrew you are using generates it. but im not 100% sure what it stands for.

With the 1.0.0 release of atmosphere, a lot has changed. You should delete every folder from your sdcard except `nintendo` & `emummc` (if you have it) and put the new DeepSea files on it. Please do not just simply override the files.
If you have a patched console or Modchip you might need a few more steps.

Jfowler30, 2021-09-27:
Oh ok I will do that, I just thought the deep sea updater was supposed to do that no worries

closed 2021-09-27 by Jfowler30

Slluxx, 2021-09-27:
> Oh ok I will do that, I just thought the deep sea updater was supposed to do that no worries

the deepsea updater is **long** gone. You must be on a version thats over a year old. Switch AiO updater has an updating mechanism for deepsea. thats what you are supposed to use.