Team-Neptune/DeepSea #74: Invalid NCA

idarkside, 2021-10-03:
When running deepsea with HOS 13 + Atmosphere 1.1.1 and booting from hekate in to emuMMC i am unable to install some NSP files as i get an invalid NCA error.

The same NSP's will not run and show as corrupt

However if i boot using fusee.bin i am able to install/run without error

Please can you advise?

hax4dazy, 2021-10-03:
`For pirated eshop-games, forwarders and other unofficial stuff you need signature patches. You can download them seperatly via the included "Switch AIO updater" homebrew. As their general purpose is to allow piracy we're not providing any help with installation or problems of said patches or pirated games afterwards.`

idarkside, 2021-10-03:
Ok no problem thanks,

Any idea why booting with fusee does not cause this issue?

hax4dazy, 2021-10-03:
I'm unsure as to why that might happen, users always are free to modify their DeepSea installation so you could have done anything with it

idarkside, 2021-10-03:
thats odd,

i have not made any further modifications, 

- i installed deepsea, booted to deepsea emuMMC, 
- downloaded sigpatches with the AIO, 
- downloaded fusee with AIO, 
- tried to install the nsp, 
- recieved the error, 
- rebooted to hekate and loaded the fusee payload, 
- installed the game fine :/ 
 
this is why i am unsure why i cant install (only some) without booting the fusee payload

hax4dazy, 2021-10-03:
You may need to edit the hekate config file to include `kip1patch=nosigchk`

idarkside, 2021-10-03:
Unfortunatly that did not fix the issue,

What is the differance between launching the emuMMC from the launch menu of hekate and booting the CFW using fusee.bin?

it still shows the Nintendo Deepsea loading logo, the sysmodules still load & the system version shows 13.0.0|AMS 1.1.1|E

hax4dazy, 2021-10-03:
There should be no difference but its better to stick to hekate seeing as its got many more features then fusese does. Did you install the "hekate" patches from AiO?

idarkside, 2021-10-03:
Sorry for the confusion you are correct, i did not add "kip1patch=nosigchk" to the correct line, i have since adjusted this and it works brilliantly.

To answer your previous question i did indeed download the hekate patches

Thank you for your help,

closed 2021-10-03 by idarkside