Team-Neptune/DeepSea #80: Major crashing

renegade2k82, 2021-11-03:
Ok so my switch is a v1 launch unit now i am using the latest version package advanced and i know my 13.1 firmware is supported well i keep getting these crashes.Now the only thing i have done to my switch was format the system and use a brand new fresh 512gb sandisk switch branded sd card.I copied all the files over and everything and made sure to eject the sd card before taking it out.I also do not use a emu nand and dont install any nsp files or anything so this should not b
[image]
e happing.Hope someone can help.

Slluxx, 2021-11-04:
Do you have any sysmodules active?
Try deleting all files except `switch`, `nintendo`, `emummc` and `config` (if you have them) and then copy files from the latest release to your sdcard.
Don't just simply overwrite the old files. 

Make sure you use the included Hekate payload and from within Hekate (under payloads -> CommonProblemResolver ) you can also disable all sysmodules.

If the problem persists, please join our discord and open a ticket.

renegade2k82, 2021-11-04:
The only sysmodules i have active are telsa and ftp light.also this was a brand new sd card and was never used so there were no old files on it.

Slluxx, 2021-11-04:
You said you "copied all the files over" so i assumed from the old sdcard, which might not be our most up to date version.
anyway, please open a ticket in discord. the issues here are regarding the builder and its modules, not for switch support.

closed 2021-11-04 by Slluxx