Team-Neptune/DeepSea #81: Crash on hbmenu after updating Atmosphère 1.2.4

WollenMoth, 2021-11-10:
Error Code: 2144-0001 (0x290)
Program: 0100000000001000
Firmware: 13.1.0 (Atmosphère 1.2.4-master-e256261b8)

I decided to update since AIO Updater doesn't work on Atmosphère 1.2.2, it says that the downloaded files are not zip files.
After updating, when I try to open hbmenu, it crashes with the error code shown above.
Also I'm having problems with the micro SD card, it says it was used in another Nintendo Switch, then delete the data, the NS reboots and the error is not shown, but if I reboot again the error is still there.

Slluxx, 2021-11-10:
> After updating, when I try to open hbmenu, it crashes with the error code shown above.

Because you updated your system with some theme(s) installed. 
- Boot to hekate
- click "payloads" 
- click "commonproblemresolver"
- click "uninstall themes" (or anything along those lines)
- then "reboot to update.bin" to get back to hekate

> Also I'm having problems with the micro SD card, it says it was used in another Nintendo Switch, then delete the data, the NS reboots and the error is not shown, but if I reboot again the error is still there.

is your sdcard exfat or fat32?

Slluxx, 2021-11-11:
inform yourself how "h2testw" works and you cancheck if your card is really fake or not. If it is, this will be the cause of your problem.

WollenMoth, 2021-11-11:
Yes, most likely it's fake.

Warning: Only 7566 of 7672 MByte tested.
The media is likely to be defective.
6.4 MByte OK (13239 sectors)
7.3 GByte DATA LOST (15481929 sectors)
Details:36.5 KByte overwritten (73 sectors)
0 KByte slightly changed (< 8 bit/sector, 0 sectors)
7.3 GByte corrupted (15481856 sectors)
36.5 KByte aliased memory (73 sectors)
First error at offset: 0x0000000000676e00
Expected: 0x0000000000676e00
Found: 0x00eacdb7c0f37e57
H2testw version 1.3
Writing speed: 8.08 MByte/s
Reading speed: 8.32 MByte/s
H2testw v1.4

Slluxx, 2021-11-11:
Either fake or broken, yeah.

closed 2021-11-29 by WollenMoth