Team-Neptune/DeepSea #84: v3.3.2 支持 OLED 硬破機嗎？

leo5579, 2021-11-17:
v3.3.2 支持 OLED 硬破機嗎？

Slluxx, 2021-11-17:
I have no idea what you want to know. Please use the english language to communicate successfully with us.

I assume you are asking if an oled switch can be modded? Only with a very expensive modchip.
We dont recomend those though.

closed 2021-11-17 by Slluxx

Exorcism0666, 2021-11-17:
I used google translate because I am not Chinese but I think he meant "does version 3.2.2 of Deepsea support Nintendo switch OLED?"

hax4dazy, 2021-11-17:
Honestly, I've got no idea if it does or not. I don't have an oled switch and I don't plan on getting one. If anyone does and they read this feel free to ping me in the [discord server](https://invite.teamneptune.net/)