Team-Neptune/DeepSea #89: Suggestion: Add NX-Activity-Log

mid-kid, 2022-01-14:
https://github.com/tallbl0nde/NX-Activity-Log/releases

Rationale:
* It's neat
* No seriously it's a pretty cool app

Slluxx, 2022-01-14:
Thanks for your suggestion.

When we are adding new homebrews to one of our packages, we always have to think about size vs the amount of people actually using the homebrew. A lot of homebrews like this are neat and deserve to be used, the truth is that very few people would actually use them. DeepSea is not made to include every single homebrew in existence but rather to be a beginner-friendly way into the scene with the most usefull/most used homebrews.

We can add this homebrew to our builder so it can be build by our bot, website and through aio-switch-updater. Adding it to the main lineup of our packages has to be discussed internally with the team.

mid-kid, 2022-01-14:
I understand. I'm mainly suggesting it because it's an app I'd certainly recommend to many people, as it fills the void left by some of the functionality the 3DS used to provide, and is a pretty cool example of useful utility homebrew. It seems to be on the big side clocking in at around 10MB, so for that I'd reject it. As for actual usage, I have no clue, honestly. I use it regularly myself, but it's hard to gauge for other people.