Team-Neptune/DeepSea #95: Question : Why do you use TinWoo when even his creator recommends another installer ?

Informartiste, 2022-04-10:
Why do you still use TinWoo when even his creator recommends another installer, DBI installer ?

I am really NOT an expert and just try to help ...

This is where he recommends DBI installer : https://github.com/mrdude2478/TinWoo/issues/16

BIG Thanks for your work !

Slluxx, 2022-04-10:
Every fork (as far as we know) of Awoo installer has something called "gay mode", which we dont like. Some of them even have jokes against sexual minoritys hidden in it. TinWoo was a good (and seemingly the only) alternative, next to forking one and removing the bullshit content.

DBI might be a good installer but its closed source so no one knows what exactly the installer does while you have it open. In a project that itself is open source as well as is build upon all the open source software it includes, DBI is not a good choice to have in it.

Informartiste, 2022-04-10:
Ok, thanks for your answer. I totally agree and understand. And sorry to disturb you on a sunny sunday

closed 2022-04-10 by Informartiste

Slluxx, 2022-04-10:
No worries, asking questions never hurts :)

KaelosBeren, 2022-04-15:
@Slluxx What's your opinion on [AtmoXL](https://github.com/dezem/AtmoXL-Titel-Installer)? It's an open-source installer and it's regularly updated.

Slluxx, 2022-04-16:
@KaelosBeren From his commits he just updates visual stuff, languages and dependencies. One new feature seems to be http indexing but thats really it, after looking through like 4 pages of commits. Atleast awoos "bad" stuff is gone.

So uh. I dont really know. TinWoo works perfectly fine and unless it breaks or some other installer brings something new to the table, i dont really see the need of switching. Ill keep an eye on AtmoXL though and will discuss it with the team. (My personal opinion might not correlate to the teams opinion) :)