Team-Neptune/DeepSea #98: TinWoo installer crashes on v4.2.0 / Atmosphere 1.3.2

jack2game, 2022-05-02:
As the title suggests. TinWoo version 1.0.5

If run in applet mode, it crashes the entire console with the standard Atmosphere BSOD.

If run in non applet mode, the game itself will be forced to close.

Had to downgrade to v4.1.0 in order to use TinWoo.

Slluxx, 2022-05-02:
You are right. We need to look into this

jack2game, 2022-06-05:
This is now fixed with the new TinWoo release.

It is not on their github release page and the hombrew can be updated using its build-in updater, otherwise you'll probably have to compile from their source.

Tested on Firmware 14.1.1 and Atmosphere 1.3.2 with Tinfoil 1.0.11

Slluxx, 2022-06-05:
I updated tinwoo yesterday from their github and it works.
we just have to compile a new deepsea update

closed 2022-06-05 by Slluxx