Team-Neptune/DeepSea #102: please update hekate 5.8.0

AquaHacker, 2022-07-23:
please update hekate 0.5.8

R3m0ved, 2022-08-07:
Also nx-hbmenu to 3.5.1 and NX-Shell 4.0.1, you can replace it yourself in your SD but anyway...
As an improvement, integrating a 'tag' or a 'version' property in the build script's settings could be nice to have for people to know which versions are bundled
https://github.com/Team-Neptune/DeepSea/blob/master/src/settings.json

Slluxx, 2022-08-08:
@R3m0ved No we do update for other reasons too. We just cant keep track of every single homebrew (just like we cant for every homebrew to update if nessesary). 

Everything in this repo is automated. Therefore we dont want to write stuff into settings.json manually. We could let the buildscript update the file, but it has to be done in the dev branch, where most people wouldnt find it anyway. If we did it in master, we would trigger a new build and since that would write into master too, we'd have a infinite cycle of loops.

We could write the versions of homebrews into the release text but fetching a homebrews latest version is an API call added to the overall call list. People who want to run the script without having a github private key will run into api limits & restrictions by github. Thats why we didnt do it yet.

We will update Hekate next week, when we have enough staff to test the new release.
(you may not know but each and every release gets tested thoroughly to ensure everything is working)

R3m0ved, 2022-08-08:
Thx for the reply !

Yep, seems fair to do some tests before any release, i've seen your checklist :)
https://github.com/Team-Neptune/DeepSea/projects/1

Did not know about the private key needed to fetch a specific tag, makes sense now

hax4dazy, 2022-10-21:
Hekate got updated in latest release

closed 2022-10-21 by hax4dazy