Team-Neptune/DeepSea #104: Idea: Add some stuff

SirGamers, 2022-08-08:
Add themezer-nx and/or NXThemes Installer to Normal and Advanced.
Add MissionControl to Normal.
Add AIO-Switch-Updater to Minimal
Switch out JKSV for Checkpoint or add both to Advanced but only one for Normal.

hax4dazy, 2022-08-08:
> Add themezer-nx and/or NXThemes Installer to Normal and Advanced.
We might decide to add some theming tool in the future.
> Add MissionControl to Normal.
No. We removed it from normal in the past to create a more varied pool of packs that people can choose from. New users can use normal but if they decide they want more they can choose to use advanced. Ontop of that nothing is different from normal to advanced besides the fact that normal doesn't have any sysmodules.
> Add AIO-Switch-Updater to Minimal
Eh? We internally decided that minimal should remain a package for people that *just* want the basics. AiO might just be a bit too bloated for a "minimal" package.
> Switch out JKSV for Checkpoint or add both to Advanced but only one for Normal.
Sorry but again no. Checkpoint at one point didn't support the latest version of atmosphere, if this happens again this means we'd either have to wait  for it or release without it (which we have done before).

I'll tag @Slluxx to see if he has something extra to add and if we want to add some theming tool.

Slluxx, 2022-08-08:
I am okay with adding aio-switch-updater to minimal, however hax is right and we left it out intentionally for exactly that reason.

Sysmodules (other than tesla) wont be added to normal.

For adding themeing stuff, we should make a community poll.

I dont see a reason why we should swap jksv and checkpoint. both do the same job and in my opinion jksv is superior. If you dont mind then please explain why you would like to have that change and we can talk about it again.

----

It would really help if you explain the reason behind your wishes. They may be valid and we should give them some serious thought. But if you dont provide it, we will just tell you that there is a reason behind how it currently is.

R3m0ved, 2022-08-18:
Can you move Emuiibo to Advanced ? Does not work with every lang atm, crashes atmo and seems a bit much for a normal bundle, adds extra stuff to atmo/contents folder

closed 2022-10-14 by SirGamers