Team-Neptune/DeepSea #112: Suggestion: Add NX-FPS + Status-Monitor-Overlay

fenpaws, 2022-11-19:
Hey there, Team Neptune. 
First off, I love your NX CFW Bundle, It's really nicely put together!

I have a small suggestion or addition tho, could you maybe add the [Status-Monitor-Overlay](https://github.com/masagrator/Status-Monitor-Overlay)? It helps to see how the switch is performing. It also has the ability with the [NX-FPS](https://github.com/masagrator/NX-FPS) addition to display the Current FPS.

I think it would make a nice addition to the overlay collection you guys already have included.

### Links to GIthub Repos:
- [NX-FPS](https://github.com/masagrator/NX-FPS)
- [Status-Monitor-Overlay](https://github.com/masagrator/Status-Monitor-Overlay)

hax4dazy, 2022-11-19:
Status monitor overlay is already in DeepSea?

fenpaws, 2022-11-19:
Yeah, i just saw that i have the i downloaded the Normal Package, not the Advanced one. I was just about to delete the issue. Sorry my bad!

fenpaws, 2022-11-19:
I just used the Advanced version and yeah you are right its already included, but sadly not NX-FPS. So i reopened the issue

hax4dazy, 2022-11-19:
I honestly don't really see any reason to add NX-FPS into Advanced to be honest. They don't support many graphics plugins and there hasn't been and demand for it in the DeepSea survey we did in our [Discord server](https://invite.teamneptune.net/). I'm going to say no and leave this one out of the AiO pack. I am however going to ask @Slluxx on their opinion.

fenpaws, 2022-11-19:
yeah i get that, i thought since you already provide the performance overlay and a OC tool then It's maybe also smart to include something to display the FPS. The performance overlay integrates NX-FPS to display them.

Slluxx, 2022-11-19:
While i am not against it per-se, i dont like the idea of including salty-nx just for the fps plugin (or in general). The last release was 2021 and to me personally it caused more issues than it helped me. Even if it works right now, there is no guarantee it will work the next update or that massagrator would even fix it. It adds so much work for us (every release!) while there is no real payoff.

Overclocking the switch shouldn't be done by looking at the fps and wanting to reach some minimum. Its not a tool for overclocking and is just dangerous if being used like that. If i can avoid someone trying to do exactly that by not including it, i will. 

If for whatever reason you need that kind of diagnostic tool after you have done your overclock, i think you should (and you probably will) understand enough to install it yourself.

fenpaws, 2022-11-19:
I fully agree with you, thanks for your input and I'm gonna close the issue now. Thanks for your time :D

closed 2022-11-19 by fenpaws