Team-Neptune/DeepSea #114: Suggestion of homebrews for advanced ( DeepSea ) .

SrPhreaK, 2023-01-27:
Good afternoon .
I used other packages here in Brazil but I intend to use DeepSea, 
but checking the existing homebrews in the package I have some suggestions of some really important ones .

NX Themes Installer 
Themezer NX
DBI
Linkalho

hax4dazy, 2023-01-27:
Hey thanks for your suggestion. 

1. Linkalho
This app has too specific of a usage, also its user base is really small so there would only be one or two people actually using the app while it is in DeepSea
2. DBI
DBI is closed soucre (Which I don't like) and its UI / UX is really ugly
3. Theme apps
We'll make a poll in our [Discord Server](https://invite.teamneptune.net/) which one people would want the most and maybe based on that decide which one we would like to put in our pack.

0DayLin, 2023-02-14:
> Good afternoon . I used other packages here in Brazil but I intend to use DeepSea, but checking the existing homebrews in the package I have some suggestions of some really important ones .
> 
> NX Themes Installer Themezer NX DBI Linkalho

https://github.com/0DayLin/DeepSea-Ultimate/releases

Slluxx, 2023-02-14:
@0DayLin Stop spreading your cheap knockoff on our github.
You are breaking our license agreement by forking, modifying and redistributing a version of deepsea. You are also including Sigpatches, which is illegal and can get your repo DMCA'd at any time.

Apart from that, you are not even using the build system and packing & releasing your fork manually like a peasant.