Team-Neptune/DeepSea #116: hekate v6.0.2

CmpHDL, 2023-03-11:
Add hekate v6.0.2 for HOS 16.0.0 Support.
Attempted to do it myself, got a yellow screen of death.

hax4dazy, 2023-03-11:
We are waiting for atmosphere to get out of pre-release. Until then we will not update DeepSea.

closed 2023-03-11 by hax4dazy

CmpHDL, 2023-03-11:
That makes sense, @hax4dazy is there a way I can upgrade it myself and still use deepsea until atmosphere gets out of pre-release status.

hax4dazy, 2023-03-11:
We will not give anyone support for upgrading their system to the pre-release version of atmosphere. Many of the sysmodules in DeepSea are still broken. Wait for a new release to happen

CmpHDL, 2023-03-14:
@hax4dazy https://github.com/Atmosphere-NX/Atmosphere/releases/tag/1.5.1

hax4dazy, 2023-03-14:
I know, there's no need to comment about it

CmpHDL, 2023-03-14:
@hax4dazy Stay humble and continue to crank out that code for free buddy. 👍