Team-Neptune/DeepSea #122: Suggestion- Add fusee to /bootlader/payloads/

Arusene-Arsene, 2023-05-11:
this would be really helpful in my opinion. there are tons of people using Hekate to load Fusee, and its not already in the /payloads folder.

Slluxx, 2023-05-11:
There is absolutely no reason to use the fusee payload though. Especially chainloading hekate > fusee makes no sense in my opinion. If you can provide some benefits though, i am open to have a discussion.

Arusene-Arsene, 2023-05-11:
ok, so Hekate is a nice toolbox to use in general. second, if i but the wrong fusee payload, ill have to turn it off, enter RCM, and do it again. however, booting the wrong hekate payload still sends you to the normal updated hekate menu. however, this is your choice, and if you decide to add it or not is not in my control. do as you like.

hax4dazy, 2023-05-11:
We're most likely not going to do it. Feel free to add it on your own but keep in mind that no support will be given to you if you boot this way.

closed 2023-05-11 by hax4dazy