Team-Neptune/DeepSea #127: Panic occured while running atmosphere.

K-el974, 2023-09-04:
Title ID: 4200000000000010
Error:	  std::abort (0xFFE)
Report saved to /atmosphere/fatal_errors/report_0ad6c83d.bin

Hello i have updated to latest DeepSea v4.7.0 and i was on emunand 16.0.3. Everything was working perfectly until i decided to upgrade my emunand to 16.1.0.
Everything else is still working (android, ubuntu, lakka) i have a 512gb sd card.
[report_0ad6c83d.zip](https://github.com/Team-Neptune/DeepSea/files/12513018/report_0ad6c83d.zip)

Slluxx, 2023-09-04:
4200000000000010 is ldn_mitm.
When you re/install DeepSea correcty, it is off by default for this reason.

Delete sd:/atmosphere/contents/4200000000000010 if you don't care about other types of controllers.
Delete sd:/atmosphere/contents/4200000000000010/flags/boot2.flag if you just want to turn it off.

closed 2023-09-04 by Slluxx

K-el974, 2023-09-04:
You solved my issue, just by deleting sd:/atmosphere/contents/4200000000000010/flags/boot2.flag.
Thanks for your time and quick response.
Have a good day!