Team-Neptune/DeepSea #138: Question about v4.10.0 & Tinwoo

sawyermade, 2024-05-28:
I was just curious why Tinwoo is not included in the newest v4.10.0 release. Is it no longer reliable or work, etc and is the recommended way now just Goldleaf? Thanks for your time and keep up the great work.

hax4dazy, 2024-05-28:
Tinwoo was removed due to the developer being incompetent and generally annoying to work with. We decided on using Goldleaf since it also included more features then just installing backups

sawyermade, 2024-05-28:
Makes sense, the program definitely didn't have a very professional feel and using the in app update would some times break it, so for the last quite a few releases of deepsea, I would just wait until you guys did an update with a version of Tinwoo that wasn't borked. Thanks for the info and Goldleaf does seem to have a more stable development and I believe works with the ns-usbloader dev, I just haven't used it in a while since you guys included Awoo then Tinwoo. Thanks again and I guess I'll close this.

closed 2024-05-28 by sawyermade