Team-Neptune/DeepSea #140: Black Screen

Khymick, 2024-08-25:
Hi there, 

I am facing trouble starting the cfw. As I am not that deep into all steps, I write down what I do:

1. I copied the whole "advanced" zip file on my sd card
2. I own an "RCMloader" - I connect it to my switch, put the dongle on right controller side and press "vol+" and "power" at the same time
3. The dongle correctly boots - I see the screen where I can select "Launch", "More Configs", "Payloads" etc.
4. When I click on "Launch", I have 3 options: 
- "CFW (Sysnand)"
- "CFW (Emummc)"
- "Block (Sysnand´)"
5. No which of the 2 CFW I try to launch, my screen stays black.

Can somebody please help me to get this fixed? Also, if possible, it would be nice to understand where the mistake is - as I think it may be an layer 8 problem.

Brioche-bit, 2024-08-27:
Hello, 

I have the same problem and even had to format my SD card because it was no longer recognized by my computer.

Do you have a solution? Even a clean intall with a blank card didn't work.

Thanks a lot :)

Slluxx, 2024-08-27:
Github issues are for problems with the build script, not usererror or other issues. Please join our discord server where we have a team of people ready to help you.

closed 2024-08-27 by Slluxx