Team-Neptune/DeepSea #142: OFW 19.0 Support

passionatemango, 2024-10-12:


hax4dazy, 2024-10-12:
Wait until atmosphere updates, then we might do an update

jeferwang, 2024-10-16:
atmosphere pre-release 1.8.0 supported

dracula99gr, 2024-10-19:
are you guys waiting for the normal release, instead of pre-release?

hax4dazy, 2024-10-19:
> are you guys waiting for the normal release, instead of pre-release?

Correct

luvingiane, 2024-10-22:
> > are you guys waiting for the normal release, instead of pre-release?
> 
> Correct

That's a nice news. just asking, how much time do you guys need to build a new version?

hax4dazy, 2024-10-22:
Whenever Atmosphere gets out of pre-release there's a chance that we'll do an update

Zonemo001, 2024-10-30:
Could you also start supporting SaltyNX, FPX, and Tinfoil? Thanks!

coopeeo, 2024-12-10:
> I'm eagerly waiting for your new update because it's what I use. Please release it quickly.

Please do not rush the developers, and this goes for any project.

coopeeo, 2024-12-10:
Hello developers, this is just a notice that Atmosphere has released the latest release of 1.8.0. Just telling you in case you guys do not know yet.

DanielC7205, 2025-01-31:
Resolved with https://github.com/Team-Neptune/DeepSea/releases/tag/v4.13.0

closed 2025-01-31 by DanielC7205